maland16/daly-bms-uart #1: Teensy to ESP8266 conversion

athulathmudali, 2021-06-01:
Hi @maland16, I am not an expert on the subjected matter, but I have been trying to use your code with my ESP8266. I saw you have used logical level shifter connected with your teensy board directly. But I am just wondering whether I could use the ESP8266 (nodemcu) without the shifter to get the same code working. If it is possible will you be able to help me on the hardware serial setup with ESP8266 ? BTW I'm **perfectly getting the serial monitor output with null values**. i guess some tweaks will be needed.

athulathmudali, 2021-06-01:
Also note that i have directly connected BMS UART Rx Tx Gnd pins with ESP8266 Tx Rx and Gnd. But not sure your library will match with the setup.

maland16, 2021-06-02:
I use a logic level shifter in my design because the Teensy's UART serial pins are only 3V tolerant, and from what I've observed the DALY BMS communicates at a 5V logic level. This logic level shifting might not be necessary in your implementation if the ESP8266's serial rx/tx pins can handle/drive 5V. 

Either way, a quick look at the ESP8266's datasheet makes it look likly that you can talk over UART to a DALY BMS, it's just about getting the configuration right. Right now my library sits on top of the Arduino Serial library and talks UART by making calls to that library. This link is established in the Daly_BMS_UART constructor which takes in a HardwareSerial object. On a teensy "Serial1" is the HardwareSerial object that corresponds to the RX1 & TX1 pins. If you're using the arduino IDE to program your ESP8266, and you're lucky, you might be able to pass in the Serial object that corresponds to the UART pins you want to use.

I'm working on a Teensy but intend on testing this library on Arduino since I think that might make it more approachable for people. I don't currently plan on expanding support beyond that, however, unless there is a real desire. 

I hope this helps, and I am going to be working on this library and improving support & documentation in the next few days/weeks.

athulathmudali, 2021-06-02:
that a valuable piece of information and really appreciate the directions you have given. I will try it again and get back to you with an update. Thanks a lot once again @maland16 .

athulathmudali, 2021-06-04:
@maland16 , I was able to change the code into my ESP32 and all are working fine and once again thanks for your direction. Also i tried to list each and every cell voltage including their respective cell number. However the code didn't go well. I have created a below constructors. 

bool getallCellVoltages(float &cellVolt, uint8_t &cellVNum); //------------------------ **.h file** 

bool Daly_BMS_UART::getallCellVoltages(float &cellVolt, uint8_t &cellVNum) //--------------------------.**cc file**
{
    this->sendCommand(COMMAND::CELL_VOLTAGES);

    if (!receiveBytes())
    {
#ifdef DALY_BMS_DEBUG
        Serial.printf("<DALY-BMS DEBUG> Receive failed, cell voltages are not avilable!\n");
#endif
        return false;
    }

   cellVolt = ??
   CellVNum = ??
    
    return true;
}

I cannot figure out how could I return the multiple cell numbers and their respective numbers out of this I have 7 cells to monitor in my BMS. Appreciate your help on this.

athulathmudali, 2021-06-06:
@maland16 I have a good news for you. I managed to implement the rest of the commands on my ESP32 board. This is including the battery cell voltages and the rest of the commands that you didn't touch. It was bit of a hassle to handle the checksum of the battery cell voltages.

maland16, 2021-06-07:
That's great to hear! If you'd like to make a pull request to add support for the other commands to this repo feel free!

athulathmudali, 2021-06-10:
@maland16 , Sure thing. I will upload the code. Some cleanups are required as usual. perfectly working the additions. Getting the fault codes was bit of a hassle as there was over 35 fault codes. :) 👍

softwarecrash, 2021-11-12:
@athulathmudali news from your work?
im interesting too, to get your work 
optional a new function to switch the fets?

softwarecrash, 2022-04-06:
now i have rewrite it to use with esp and arduino like chips

closed 2022-04-27 by maland16