maland16/daly-bms-uart #6: Any chance of support for STATUS_INFO from the Daly?

ASMotionLab, 2022-04-16:
Hi,
Your library is great, thank you for putting in the work!
It's enabled me to build a Daly to Sofar CANBUS bridge (SMA Canbus protocol, so other inverters too). It also has MQTT client for sending BMS data to Home Assistant etc etc. Soon to be up on GitHub.
Anyway!
I noticed you haven't expanded the STATUS_INFO and DISCHARGE_CHARGE_MOS_STATUS functionality in your code, is there a reason why?
I was hoping to get the cycles from the BMS and the Battery pack capacity to pass it to the inverter & MQTT.
I appreciate you might not be maintaining this so I might give it a crack myself otherwise (but I'm not a very good programmer!)
Thanks!

maland16, 2022-04-24:
Glad you were able to make use of the library! Another community member has a big PR that's merging hopefully soon that I'm pretty sure will add the functionality you're looking for, as well as a ton of other stuff. I just have to get it cleaned up.

softwarecrash, 2022-04-24:
> Hi, Your library is great, thank you for putting in the work! It's enabled me to build a ESP32 based Daly to Sofar CANBUS bridge (SMA Canbus protocol, so other inverters too). It also has MQTT client for sending BMS data to Home Assistant etc etc. Soon to be up on GitHub. Anyway! I noticed you haven't expanded the STATUS_INFO and DISCHARGE_CHARGE_MOS_STATUS functionality in your code, is there a reason why? I was hoping to get the cycles from the BMS and the Battery pack capacity to pass it to the inverter & MQTT. I appreciate you might not be maintaining this so I might give it a crack myself otherwise (but I'm not a very good programmer!) Thanks!

hey, when it helps you can use parts of my other project (based on this work and the pr) 
https://github.com/softwarecrash/DALY-BMS-to-MQTT

when you need asap the status info and other, you can use temporary my fork with the changes until @maland16 find the time to merge it, it have all known command features.
https://github.com/softwarecrash/daly-bms-uart

ASMotionLab, 2022-04-24:
Thank you for your responses guys. Great work @softwarecrash , I'm looking forward to the new features: having control of the mosfets is pretty cool too, I'll try to add this to my project too when it's merged in 👍👍👍

closed 2023-05-15 by maland16