maland16/daly-bms-uart #18: Powering off the DALY BMS

ril3y, 2022-10-22:
Hey all, 

I didn't see a group/discord/email so this is kind of a question vs an actual "Issue".
I am working on a golf cart that is using the 250A Daly BMS.  Without getting into too much detail, my plan is to have the BMS on most of the time.  However, I would like the ability to "turn off" the bms. IE stop producing 48V output.  Is this possible to turn off the BMS over uart?  I know you can start it via a jumper wire.  I also know there is a "timer" function that if it the current is below a certain value, and the timer has expired then it will shut off.  

Anyhow thanks for putting all of this out here for everyone.

maland16, 2022-11-01:
If I recall correctly in the PC software and in the mobile app there's a way to "turn on" or "turn off" the BMS by switching on/off the charge/discharge MOSFETs. It looks like @softwarecrash implemented some functions that send those same commands across UART. See: https://github.com/maland16/daly-bms-uart/blob/main/daly-bms-uart.cpp#L418
My guess would be that turning on/off the fets achieves the functionality you're looking for. That being said, it's important to note that disabling the FETs is NOT the same as physically breaking the circuit between the battery and whatever you're powering (like a motor controller for a golf cart). It's still important to have a physical disconnect in the form of a contactor or switch for most applications.
Hope this helps!

ril3y, 2022-11-02:
I will do some testing thanks!  What I am trying to accomplish is to remove the quiescent current draw from the system.  I don't want the BMS to slowly kill the battery of the golf cart by constantly providing 48v output all of the time. I have an MCU that can send the mosfet discharge command when certain conditions (timers etc) are met. I have a contactor switch that still has to be closed to power the motors.

softwarecrash, 2022-11-02:
You can Turn Off the Out of the BMS via fet Switch.
There is No way to Turn Off the BMS self, it will going to sleep after a time you can Set. The current in sleep Mode ist verry verry Low, in my Test citcuit i have sleeping BMS in a single 18650 Zell Array over month without signifikant Draw.

ril3y, 2022-11-02:
Thanks for the information.  The 280Ah 48v battery I think will be ok then :)  The fet switch you referring to is this right?
https://diysolarforum.com/threads/daly-24v-8s-initiation-and-software.18644/#post-216020

Where you jumper a wire to a pin on the lightboard?

softwarecrash, 2022-11-02:
the fet switch is a function of this libray. not a hardware button.
take a look at the example 
  /**
   * Advanced functions:
   * bms.setBmsReset(); //Reseting the BMS, after reboot the MOS Gates are enabled!
   * bms.setDischargeMOS(true); Switches on the discharge Gate
   * bms.setDischargeMOS(false); Switches off thedischarge Gate
   * bms.setChargeMOS(true); Switches on the charge Gate
   * bms.setChargeMOS(false); Switches off the charge Gate
   */

ril3y, 2022-11-02:
Ahh I see.  I mis-read your last post.  Thanks.  I think I saw in the software that you could set the "sleep current" where if it drops below this value the BMS will sleep.  So if I am understanding this correctly the sleep time out (also settable) will occur when the current drops below the set "sleep current" for greater than the "sleep timeout" value. 

I looked around in the code but didnt see if setting these values were implemented.  I can poke around and do a pull request if they are not once I get them done.

softwarecrash, 2022-11-02:
so i remember a sleep current not exsist by daly, only a sleep time, that mean the time in second when the bms going to sleep when no current is going on, the sleep time can you set by the app or pc software.

ril3y, 2022-11-03:
Hmm. I could have sworn you could set it but maybe not.  Seems like if you have an accessorie drawing the smallest bit of current it will just always stay on.  Thanks I will double check on this.

softwarecrash, 2022-11-03:
> Hmm. I could have sworn you could set it but maybe not. Seems like if you have an accessorie drawing the smallest bit of current it will just always stay on. Thanks I will double check on this.

yes, when a smal current is avaible, the bms will stay awake, 0,5A is enugh.
you can build a smal programm to disable the output by esp (notice that a arduino have 5v pins, the bms is not 5v tollerant, you need a level shifter) and wake the bms again by the wake button and then enable the output again.

ril3y, 2022-11-03:
When you say 
"you can build a smal programm to disable the output by esp"
What do you mean?  I am specifically trying to do this with a board I am developing :)
Do you mean send the discharge fets command.. Then the bms will time out and shut down completely?
Then I would need to use the gpio to bring it back online?  That is exactly what I am looking for.
If I understand correctly :)

softwarecrash, 2022-11-03:
> When you say "you can build a smal programm to disable the output by esp" What do you mean? I am specifically trying to do this with a board I am developing :) Do you mean send the discharge fets [...]

yes, when i remember it right, when you send the dischargefet off command, no output will comes from the bms, so no uccrent, the bms will going to sleep. with the wake up button from light bar or uart port (it have one too) you can bring back the bms, then switch on the dischargefet

ril3y, 2022-11-03:
1. Oh ok great!  Just to recap.  
2. You discharge the mosfets via UART.  This will cut current output to zero.  
3. At this point the BMS counter starts.  
4. Once this counter has expired.  The BMS fully shuts down to the point of needing to send a GPIO signal into the light board port.

NOTE: If you want to turn the bms back "on" and the timer has not expired you can send a UART command to charge the mosfets.  In which the timer would reset as the current is flowing.

maland16, 2023-05-15:
I'll probably drop a link to this in the README since it's useful knowledge, thanks @ril3y !

closed 2023-05-15 by maland16

apaczenko1993, 2024-10-22:
So I will have a question. What should I send via UART to the BMS to wake it up? After some time of inactivity, the BMS probably goes into sleep mode and does not respond to commands until I start discharging/charging it. Is there any command that will wake him up?

softwarecrash, 2024-10-22:
> So I will have a question. What should I send via UART to the BMS to wake it up? After some time of inactivity, the BMS probably goes into sleep mode and does not respond to commands until I start discharging/charging it. Is there any command that will wake him up?

there is no comand for wake up in this case, when BMS sleep the comunication on BMS side sleep too. only way is to wake it with the wake pin or charge/discharge

apaczenko1993, 2024-10-22:
> there is no comand for wake up in this case, when BMS sleep the comunication on BMS side sleep too. only way is to wake it with the wake pin or charge/discharge

Can you tell more about wake pin? :)

softwarecrash, 2024-10-22:
> Can you tell more about wake pin? :)

[image]