maland16/daly-bms-uart #19: Setting SoC

sebdehne, 2022-12-13:
I figured it is also possible to set SoC with the 0x21 command (RTC is also writted at the same time):

A5 40 21 08 - the usual header
160C0D151D27 - 6 bytes for the date/time. YY-MM-DD HH:MM:SS. This example 12/13/2022 9:29:39
01 2C - SoC: 300/10 = 30%)
C3

softwarecrash, 2022-12-29:
thanks, i have found the 21 command for months and implement it, but dont know about the time, not sure that the bms need a time. take a look at my PR, i think thats all correct for setting SOC

maland16, 2023-05-15:
I'm gonna try and pull that PR in soon

197Fabian, 2023-11-12:
Hi,

I also want to set the SOC of the DALY BMS with the esp32, But I don’t know how.
Could you please help me?
Do you have any example how I must Realize that??

nschermer, 2024-05-23:
Changing the SOC is fairly simple:

```
uint16_t socScaled = soc * 10; // soc is a number between 0 and 100
this->my_txBuffer[10] = highByte(socScaled);
this->my_txBuffer[11] = lowByte(socScaled);
this->sendCommand(0x21);

// Clear the buffer for further use
this->my_txBuffer[10] = 0x00;
this->my_txBuffer[11] = 0x00;
```

All the registers before 10 are for a date/time format, not really interesting and if you keep them 0 the daly works fine.