maland16/daly-bms-uart #20: Add serial port definition Esp32 support

he77PC011, 2022-12-28:
Add serial port definition （ ----Daly_BMS_UART bms(Serial2, 16, 17);--- ）
Adding esp32 support

maland16, 2023-05-15:
Not sure what you mean, could you elaborate?

he77PC011, 2023-05-23:
Arduino hardware   esp32 
#define BMS_SERIAL Serial2 

02:29:54.276 -> Basic BMS Data:              49.40V 0.00I 84.50% 
02:29:54.276 -> Package Temperature (C):     0.00
02:29:54.276 -> Highest Cell Voltage:        #0 with voltage 0.00
02:29:54.276 -> Lowest Cell Voltage:         #0 with voltage 0.00
02:29:54.276 -> Number of Cells:             0
02:29:54.276 -> Number of Temp Sensors:      0
02:29:54.276 -> BMS Chrg / Dischrg Cycles:   0
02:29:54.276 -> BMS Heartbeat:               0
02:29:54.310 -> Discharge MOSFet Status:     0
02:29:54.310 -> Charge MOSFet Status:        0
02:29:54.310 -> Remaining Capacity mAh:      0
02:29:54.310 -> Level one Cell V to High:    0
02:29:54.310 -> Press any key and hit enter to query data from the BMS...
02:29:55.360 -> Basic BMS Data:              49.40V 0.00I 84.50% 
02:29:55.360 -> Package Temperature (C):     0.00
02:29:55.360 -> Highest Cell Voltage:        #0 with voltage 0.00
02:29:55.360 -> Lowest Cell Voltage:         #0 with voltage 0.00
02:29:55.360 -> Number of Cells:             0
02:29:55.360 -> Number of Temp Sensors:      0
02:29:55.360 -> BMS Chrg / Dischrg Cycles:   0
02:29:55.360 -> BMS Heartbeat:               0
02:29:55.360 -> Discharge MOSFet Status:     0
02:29:55.360 -> Charge MOSFet Status:        0
02:29:55.394 -> Remaining Capacity mAh:      0
02:29:55.394 -> Level one Cell V to High:    0
02:29:55.394 -> Press any key and hit enter to query data from the BMS...

maland16, 2023-06-06:
It looks like you're receiving the basic data (Voltage, Current, and SOC) but no other data. Given that, I think it's likely your BMS isn't responding to those other commands, or is responding with all 0s. Maybe it's an older version that doesn't implement those commands. Would you mind sharing a picture of what your BMS looks like. It might also be helpful to define the debug tag in the library so that all the debug info is printed out.