maland16/daly-bms-uart #22: Received data not looking correct

stuartsjg, 2023-05-01:
Hi, just trying to set this up - very excited to make better use of the BMS.   I am using an ESP32 and changed Serial1 to Serial2 (the board i have Serial1 is attached to an OLED display), comms established based on checking the lines on storage scope (logic analyser on order!) but it looks like what is coming back does is not recognoised or is a default perhaps:

`02:00:36.743 -> Press any key to query data from the BMS
02:00:36.743 -> basic BMS Data:              0.10V -3000.00I 0.00% 
02:00:36.743 -> Package Temperature:         -39.00
02:00:36.743 -> Highest Cell Voltage Nr:     1 with voltage 0.00
02:00:36.743 -> Lowest Cell Voltage Nr:      0 with voltage 65.28
02:00:36.788 -> Number of Cells:             0
02:00:36.788 -> Number of Temp Sensors:      0
02:00:36.788 -> BMS Chrg / Dischrg Cycles:   0
02:00:36.788 -> BMS Heartbeat:               0
02:00:36.788 -> Discharge MOSFet Status:     0
02:00:36.788 -> Charge MOSFet Status:        0
02:00:36.788 -> Remaining Capacity mAh:      0
02:00:36.788 -> Level one Cell V to High:    0
`

Is this an issue with the library or BMS firmware, or just my implementation.    The BMS works fine with the supplied Bluetooth dongle.

Thanks,
Stuart

maland16, 2023-05-15:
That's odd, it seems like the read operations are succeeding (or at least some of them are) but the data returned is no good. Would you mind turning on debug printing in the library by uncommenting this #define https://github.com/maland16/daly-bms-uart/blob/main/daly-bms-uart.cpp#L5 and copying the output here?
  
I'd also recommend double checking your connections, not only that you have the right pins connected but the connections are solid. A logic analyzer would be super useful for debugging your setup, but even a normal oscilloscope would be useful to see if the signals are at the right level and the lines don't look too noisey.

stuartsjg, 2023-05-27:
Hi - Sorry i never had a chance to follow-up, the BMS developed a fault and i received a refund and bought a Seplos BMS which was plug and play with my inverter on Pylontech mode.

Many thanks,
Stuart

closed 2023-06-06 by maland16

rarrr, 2023-11-05:
For future people that get the same thing eg something like this:
{'chargeperc': 0.0, 'current': -3000.0, 'voltage': 0.0}
It means the BMS is a sleep and a quick charge will wake it up.