maland16/daly-bms-uart #25: Ability to limit output current or power

ss89, 2023-10-14:
I'm planning on building a battery system with daly bms and thus i don't know yet what can be done on bms side.

however what would be great is the ability to set the max output current or power.

that is to be able to slow start (gradually increase current on a slow ramp) the output, e.g. when connected to a grid tie inverter to not fry the caps.

maland16, 2023-10-14:
I don't believe the DALY BMS's have the ability to actively limit current, I think they can only sense current and disable themselves if the current exceeds the setpoint.

ss89, 2023-10-15:
You are right! I just took some time to watch a youtube video about the app and i couldn't find a functionality like i described.
Neither could i find docs on their homepage.

i'll close the issue then.

closed 2023-10-15 by ss89