maland16/daly-bms-uart #31: Checksum fails while charging

ctaskin, 2025-02-04:
Hi,
I noticed that I am getting alot of checksum failed and "Received the wrong number of bytes! Expected 13, got 3" error when I start the charger, but I never get the error when charger is disabled.
I am pasting logs for both cases. Do you have any idea about the reason? Could it be some kind of noise related to my cables etc? I tried with other BMS and having the same issue..
Thanks

`

---------------------CHARGER ENABLED---------------------------

12:47:41.724 > <DALY-BMS DEBUG> Calculated checksum: 91, Received: 91
12:47:41.724 > <DALY-BMS DEBUG> Send command: 0x94 Checksum = 0x81
12:47:41.774 > <DALY-BMS DEBUG> Calculated checksum: 140, Received: 142
12:47:41.774 > <DALY-BMS DEBUG> **Error: Checksum failed!**
12:47:41.774 > <DALY-BMS DEBUG> RX Buffer: [,0x165,0x1,0x148,0x8,0x4,0x0,0x0,0x0,0x0,0x0,0x1,0x69,0x142]
12:47:41.774 > <DALY-BMS DEBUG> **Receive failed, Status info won't be modified!**
12:47:42.525 > <DALY-BMS DEBUG> Send command: 0x90 Checksum = 0x7D
12:47:42.574 > <DALY-BMS DEBUG> Calculated checksum: 80, Received: 80
12:47:42.574 > <DALY-BMS DEBUG> 13.10V, 0.00A, 23.40SOC
12:47:42.574 > <DALY-BMS DEBUG> Send command: 0x91 Checksum = 0x7E
12:47:42.624 > <DALY-BMS DEBUG> Calculated checksum: 17, Received: 17
12:47:42.624 > <DALY-BMS DEBUG> Send command: 0x92 Checksum = 0x7F
12:47:42.674 > <DALY-BMS DEBUG> Calculated checksum: 186, Received: 186
12:47:42.674 > <DALY-BMS DEBUG> Send command: 0x93 Checksum = 0x80
12:47:42.724 > <DALY-BMS DEBUG> Calculated checksum: 92, Received: 92
12:47:42.724 > <DALY-BMS DEBUG> Send command: 0x94 Checksum = 0x81
12:47:42.774 > <DALY-BMS DEBUG> Calculated checksum: 142, Received: 142
12:47:42.774 > <DALY-BMS DEBUG> Send command: 0x95 Checksum = 0x82
12:47:42.824 > <DALY-BMS DEBUG> Calculated checksum: 1, Received: 1
12:47:42.824 > <DALY-BMS DEBUG> Frame No.: 1 Cell No: 1. 3282mV
12:47:42.824 > <DALY-BMS DEBUG> Frame No.: 1 Cell No: 2. 3300mV
12:47:42.824 > <DALY-BMS DEBUG> Frame No.: 1 Cell No: 3. 3299mV
12:47:42.837 > <DALY-BMS DEBUG> Calculated checksum: 21, Received: 21
12:47:42.838 > <DALY-BMS DEBUG> Frame No.: 2 Cell No: 4. 3301mV
12:47:42.838 > <DALY-BMS DEBUG> Send command: 0x96 Checksum = 0x83
12:47:42.894 > <DALY-BMS DEBUG> Calculated checksum: 193, Received: 193
12:47:42.894 > <DALY-BMS DEBUG> Frame No.: 1 Sensor No: 1. 22°C
12:47:42.894 > <DALY-BMS DEBUG> Send command: 0x97 Checksum = 0x84
12:47:42.944 > <DALY-BMS DEBUG> Calculated checksum: 69, Received: 69
12:47:42.944 > <DALY-BMS DEBUG> Cell Balance State: 0000
12:47:42.944 > <DALY-BMS DEBUG> Send command: 0x98 Checksum = 0x85
12:47:42.994 > <DALY-BMS DEBUG> Calculated checksum: 70, Received: 70
12:47:43.525 > <DALY-BMS DEBUG> Send command: 0x90 Checksum = 0x7D
12:47:43.574 > <DALY-BMS DEBUG> Calculated checksum: 84, Received: 84
12:47:43.574 > <DALY-BMS DEBUG> 13.20V, 0.30A, 23.40SOC
12:47:43.574 > <DALY-BMS DEBUG> Send command: 0x91 Checksum = 0x7E
12:47:43.624 > <DALY-BMS DEBUG> Calculated checksum: 42, Received: 42
12:47:43.624 > <DALY-BMS DEBUG> Send command: 0x92 Checksum = 0x7F
12:47:43.674 > <DALY-BMS DEBUG> Calculated checksum: 202, Received: 186
12:47:43.674 > <DALY-BMS DEBUG> **Error: Checksum failed!**
12:47:43.674 > <DALY-BMS DEBUG> RX Buffer: [,0x165,0x1,0x146,0x24,0x62,0x1,0x62,0x1,0x255,0x255,0x255,0x255,0x186]
12:47:43.674 > <DALY-BMS DEBUG> Receive failed, Temp values won't be modified!
12:47:44.525 > <DALY-BMS DEBUG> Send command: 0x90 Checksum = 0x7D
12:47:44.574 > <DALY-BMS DEBUG> Calculated checksum: 78, Received: 78
12:47:44.574 > <DALY-BMS DEBUG> 13.20V, -0.30A, 23.40SOC
12:47:44.574 > <DALY-BMS DEBUG> Send command: 0x91 Checksum = 0x7E
12:47:44.624 > <DALY-BMS DEBUG> Calculated checksum: 38, Received: 38
12:47:44.624 > <DALY-BMS DEBUG> Send command: 0x92 Checksum = 0x7F
12:47:44.674 > <DALY-BMS DEBUG> Calculated checksum: 186, Received: 186
12:47:44.674 > <DALY-BMS DEBUG> Send command: 0x93 Checksum = 0x80
12:47:44.724 > <DALY-BMS DEBUG> Calculated checksum: 95, Received: 95
12:47:44.724 > <DALY-BMS DEBUG> Send command: 0x94 Checksum = 0x81
12:47:44.774 > <DALY-BMS DEBUG> Calculated checksum: 142, Received: 142
12:47:44.774 > <DALY-BMS DEBUG> Send command: 0x95 Checksum = 0x82
12:47:44.824 > <DALY-BMS DEBUG> Calculated checksum: 25, Received: 25
12:47:44.824 > <DALY-BMS DEBUG> Frame No.: 1 Cell No: 1. 3303mV
12:47:44.824 > <DALY-BMS DEBUG> Frame No.: 1 Cell No: 2. 3301mV
12:47:44.824 > <DALY-BMS DEBUG> Frame No.: 1 Cell No: 3. 3301mV
12:47:44.838 > <DALY-BMS DEBUG> Calculated checksum: 28, Received: 28
12:47:44.838 > <DALY-BMS DEBUG> Frame No.: 2 Cell No: 4. 3305mV
12:47:44.838 > <DALY-BMS DEBUG> Send command: 0x96 Checksum = 0x83
[... 738 lines cut ...]
12:49:21.726 > <DALY-BMS DEBUG> Send command: 0x94 Checksum = 0x81
12:49:21.776 > <DALY-BMS DEBUG> Calculated checksum: 142, Received: 142
12:49:21.776 > <DALY-BMS DEBUG> Send command: 0x95 Checksum = 0x82
12:49:21.826 > <DALY-BMS DEBUG> Calculated checksum: 20, Received: 20
12:49:21.826 > <DALY-BMS DEBUG> Frame No.: 1 Cell No: 1. 3294mV
12:49:21.826 > <DALY-BMS DEBUG> Frame No.: 1 Cell No: 2. 3301mV
12:49:21.826 > <DALY-BMS DEBUG> Frame No.: 1 Cell No: 3. 3305mV
12:49:21.839 > <DALY-BMS DEBUG> Calculated checksum: 34, Received: 34
12:49:21.840 > <DALY-BMS DEBUG> Frame No.: 2 Cell No: 4. 3307mV
12:49:21.840 > <DALY-BMS DEBUG> Send command: 0x96 Checksum = 0x83
12:49:21.896 > <DALY-BMS DEBUG> Calculated checksum: 193, Received: 193
12:49:21.896 > <DALY-BMS DEBUG> Frame No.: 1 Sensor No: 1. 22°C
12:49:21.896 > <DALY-BMS DEBUG> Send command: 0x97 Checksum = 0x84
12:49:21.946 > <DALY-BMS DEBUG> Calculated checksum: 70, Received: 70
12:49:21.946 > <DALY-BMS DEBUG> Cell Balance State: 0000
12:49:21.946 > <DALY-BMS DEBUG> Send command: 0x98 Checksum = 0x85
12:49:21.996 > <DALY-BMS DEBUG> Calculated checksum: 70, Received: 70
12:49:22.527 > <DALY-BMS DEBUG> Send command: 0x90 Checksum = 0x7D
12:49:22.576 > <DALY-BMS DEBUG> Calculated checksum: 81, Received: 81
12:49:22.576 > <DALY-BMS DEBUG> 13.20V, -0.30A, 23.70SOC
12:49:22.576 > <DALY-BMS DEBUG> Send command: 0x91 Checksum = 0x7E
12:49:22.626 > <DALY-BMS DEBUG> Calculated checksum: 30, Received: 30
12:49:22.626 > <DALY-BMS DEBUG> Send command: 0x92 Checksum = 0x7F

`

ctaskin, 2025-02-04:
-------- DISCHARGING 10Amp-------- NO ERRORS

13:44:19.839 > <DALY-BMS DEBUG> Frame No.: 1 Cell No: 1. 3248mV
13:44:19.839 > <DALY-BMS DEBUG> Frame No.: 1 Cell No: 2. 3259mV
13:44:19.839 > <DALY-BMS DEBUG> Frame No.: 1 Cell No: 3. 3252mV
13:44:19.852 > <DALY-BMS DEBUG> Calculated checksum: 131, Received: 131
13:44:19.853 > <DALY-BMS DEBUG> Frame No.: 2 Cell No: 4. 3243mV
13:44:19.853 > <DALY-BMS DEBUG> Send command: 0x96 Checksum = 0x83
13:44:19.909 > <DALY-BMS DEBUG> Calculated checksum: 193, Received: 193
13:44:19.909 > <DALY-BMS DEBUG> Frame No.: 1 Sensor No: 1. 22°C
13:44:19.909 > <DALY-BMS DEBUG> Send command: 0x97 Checksum = 0x84
13:44:19.959 > <DALY-BMS DEBUG> Calculated checksum: 71, Received: 71
13:44:19.959 > <DALY-BMS DEBUG> Cell Balance State: 0000
13:44:19.959 > <DALY-BMS DEBUG> Send command: 0x98 Checksum = 0x85
13:44:20.009 > <DALY-BMS DEBUG> Calculated checksum: 198, Received: 198
13:44:20.538 > <DALY-BMS DEBUG> Send command: 0x90 Checksum = 0x7D
13:44:20.589 > <DALY-BMS DEBUG> Calculated checksum: 251, Received: 251
13:44:20.589 > <DALY-BMS DEBUG> 13.00V, -10.50A, 0.00SOC
13:44:20.589 > <DALY-BMS DEBUG> Send command: 0x91 Checksum = 0x7E
13:44:20.639 > <DALY-BMS DEBUG> Calculated checksum: 193, Received: 193
13:44:20.639 > <DALY-BMS DEBUG> Send command: 0x92 Checksum = 0x7F
13:44:20.689 > <DALY-BMS DEBUG> Calculated checksum: 186, Received: 186
13:44:20.689 > <DALY-BMS DEBUG> Send command: 0x93 Checksum = 0x80
13:44:20.739 > <DALY-BMS DEBUG> Calculated checksum: 194, Received: 194
13:44:20.739 > <DALY-BMS DEBUG> Send command: 0x94 Checksum = 0x81
13:44:20.789 > <DALY-BMS DEBUG> Calculated checksum: 140, Received: 140
13:44:20.789 > <DALY-BMS DEBUG> Send command: 0x95 Checksum = 0x82
13:44:20.839 > <DALY-BMS DEBUG> Calculated checksum: 135, Received: 135
13:44:20.839 > <DALY-BMS DEBUG> Frame No.: 1 Cell No: 1. 3248mV
13:44:20.839 > <DALY-BMS DEBUG> Frame No.: 1 Cell No: 2. 3259mV
13:44:20.839 > <DALY-BMS DEBUG> Frame No.: 1 Cell No: 3. 3252mV
13:44:20.853 > <DALY-BMS DEBUG> Calculated checksum: 131, Received: 131
13:44:20.853 > <DALY-BMS DEBUG> Frame No.: 2 Cell No: 4. 3243mV
13:44:20.853 > <DALY-BMS DEBUG> Send command: 0x96 Checksum = 0x83
13:44:20.909 > <DALY-BMS DEBUG> Calculated checksum: 193, Received: 193
13:44:20.909 > <DALY-BMS DEBUG> Frame No.: 1 Sensor No: 1. 22°C
13:44:20.909 > <DALY-BMS DEBUG> Send command: 0x97 Checksum = 0x84
13:44:20.959 > <DALY-BMS DEBUG> Calculated checksum: 71, Received: 71
13:44:20.959 > <DALY-BMS DEBUG> Cell Balance State: 0000
13:44:20.959 > <DALY-BMS DEBUG> Send command: 0x98 Checksum = 0x85
13:44:21.009 > <DALY-BMS DEBUG> Calculated checksum: 198, Received: 198
13:44:21.538 > <DALY-BMS DEBUG> Send command: 0x90 Checksum = 0x7D
13:44:21.589 > <DALY-BMS DEBUG> Calculated checksum: 5, Received: 5
13:44:21.589 > <DALY-BMS DEBUG> 13.00V, -9.50A, 0.00SOC
13:44:21.589 > <DALY-BMS DEBUG> Send command: 0x91 Checksum = 0x7E
13:44:21.639 > <DALY-BMS DEBUG> Calculated checksum: 202, Received: 202
13:44:21.639 > <DALY-BMS DEBUG> Send command: 0x92 Checksum = 0x7F
13:44:21.689 > <DALY-BMS DEBUG> Calculated checksum: 186, Received: 186
13:44:21.689 > <DALY-BMS DEBUG> Send command: 0x93 Checksum = 0x80
13:44:21.739 > <DALY-BMS DEBUG> Calculated checksum: 195, Received: 195
13:44:21.739 > <DALY-BMS DEBUG> Send command: 0x94 Checksum = 0x81
13:44:21.789 > <DALY-BMS DEBUG> Calculated checksum: 140, Received: 140
13:44:21.789 > <DALY-BMS DEBUG> Send command: 0x95 Checksum = 0x82
13:44:21.839 > <DALY-BMS DEBUG> Calculated checksum: 147, Received: 147
13:44:21.839 > <DALY-BMS DEBUG> Frame No.: 1 Cell No: 1. 3255mV
13:44:21.839 > <DALY-BMS DEBUG> Frame No.: 1 Cell No: 2. 3261mV
13:44:21.839 > <DALY-BMS DEBUG> Frame No.: 1 Cell No: 3. 3255mV
13:44:21.853 > <DALY-BMS DEBUG> Calculated checksum: 141, Received: 141
13:44:21.853 > <DALY-BMS DEBUG> Frame No.: 2 Cell No: 4. 3248mV
13:44:21.853 > <DALY-BMS DEBUG> Send command: 0x96 Checksum = 0x83
13:44:21.909 > <DALY-BMS DEBUG> Calculated checksum: 193, Received: 193
13:44:21.909 > <DALY-BMS DEBUG> Frame No.: 1 Sensor No: 1. 22°C
13:44:21.909 > <DALY-BMS DEBUG> Send command: 0x97 Checksum = 0x84
13:44:21.959 > <DALY-BMS DEBUG> Calculated checksum: 69, Received: 69
13:44:21.959 > <DALY-BMS DEBUG> Cell Balance State: 0000
13:44:21.959 > <DALY-BMS DEBUG> Send command: 0x98 Checksum = 0x85
13:44:22.009 > <DALY-BMS DEBUG> Calculated checksum: 198, Received: 198
13:44:22.538 > <DALY-BMS DEBUG> Send command: 0x90 Checksum = 0x7D
13:44:22.589 > <DALY-BMS DEBUG> Calculated checksum: 11, Received: 11
13:44:22.589 > <DALY-BMS DEBUG> 13.00V, -8.90A, 0.00SOC
[... 511 lines cut ...]
13:44:41.909 > <DALY-BMS DEBUG> Frame No.: 1 Sensor No: 1. 22°C
13:44:41.909 > <DALY-BMS DEBUG> Send command: 0x97 Checksum = 0x84
13:44:41.959 > <DALY-BMS DEBUG> Calculated checksum: 69, Received: 69
13:44:41.959 > <DALY-BMS DEBUG> Cell Balance State: 0000
13:44:41.959 > <DALY-BMS DEBUG> Send command: 0x98 Checksum = 0x85
13:44:42.009 > <DALY-BMS DEBUG> Calculated checksum: 198, Received: 198
13:44:42.538 > <DALY-BMS DEBUG> Send command: 0x90 Checksum = 0x7D
13:44:42.589 > <DALY-BMS DEBUG> Calculated checksum: 17, Received: 17
13:44:42.589 > <DALY-BMS DEBUG> 13.00V, -8.30A, 0.00SOC
13:44:42.589 > <DALY-BMS DEBUG> Send command: 0x91 Checksum = 0x7E
13:44:42.639 > <DALY-BMS DEBUG> Calculated checksum: 195, Received: 195
13:44:42.639 > <DALY-BMS DEBUG> Send command: 0x92 Checksum = 0x7F
13:44:42.689 > <DALY-BMS DEBUG> Calculated checksum: 186, Received: 186
[... cut ...]
13:44:42.789 > <DALY-BMS DEBUG> Send command: 0x95 Checksum = 0x82
13:44:42.839 > <DALY-BMS DEBUG> Calculated checksum: 137, Received: 137
13:44:42.839 > <DALY-BMS DEBUG> Frame No.: 1 Cell No: 1. 3250mV
13:44:42.839 > <DALY-BMS DEBUG> Frame No.: 1 Cell No: 2. 3259mV
13:44:42.839 > <DALY-BMS DEBUG> Frame No.: 1 Cell No: 3. 3252mV
13:44:42.853 > <DALY-BMS DEBUG> Calculated checksum: 133, Received: 133
13:44:42.853 > <DALY-BMS DEBUG> Frame No.: 2 Cell No: 4. 3245mV
13:44:42.853 > <DALY-BMS DEBUG> Send command: 0x96 Checksum = 0x83

ctaskin, 2025-02-17:
Any suggestions?

maland16, 2025-02-25:
I haven't worked on this code/with this hardware in a while, but from the logs you posted it seems like it might be a case of some sort of interference/noise on the line. 

The highlighted buffer values below in particular make me think the interference is causing the communication line to get pulled high or low while charging. If you have an oscilloscope or a logic analyzer I'd recommend probing the TX and RX pins to see if there's any noticeable distortion or interference.

> 13:32:03.937 > Frame No.: 1 Sensor No: 1. 22°C
13:32:03.937 > Send command: 0x97 Checksum = 0x84
13:32:03.985 > Calculated checksum: 72, Received: 0
13:32:03.985 > Error: Checksum failed!
13:32:03.985 > RX Buffer: [,0x255,0x165,0x1,0x151,0x8,0x0,0x4**,0x0,0x0,0x0,0x0,0x0,0x0]**
13:32:03.986 > Receive failed, Cell Balance State won't be modified!
13:32:03.986 >
13:32:04.567 > Send command: 0x90 Checksum = 0x7D
13:32:04.614 > Calculated checksum: 9, Received: 0
13:32:04.615 > Error: Checksum failed!
13:32:04.615 > RX Buffer: [,0x255,0x255,0x165,0x1,0x144,0x8,0x0,0x5,0x0,0x0,0x117,0x83,0x0]
13:32:04.615 > Receive failed, V, I, & SOC values won't be modified!
13:32:05.566 > Send command: 0x90 Checksum = 0x7D
13:32:05.615 > Calculated checksum: 138, Received: 238
13:32:05.616 > Error: Checksum failed!
13:32:05.616 > RX Buffer: [,0x255,0x165,0x1,0x144,0x8,0x0,0x133,0x0,0x0,0x117,0x83,0x0,0x238]
13:32:05.616 > Receive failed, V, I, & SOC values won't be modified!
13:32:06.566 > Send command: 0x90 Checksum = 0x7D
13:32:06.612 > Calculated checksum: 12, Received: 117
13:32:06.612 > Error: Checksum failed!
13:32:06.612 > RX Buffer: [,0x255,0x255,0x255,0x255,0x43,0x11,0x130,0x65,0x2,0x21,0x0,0x0,0x117]
13:32:06.612 > Receive failed, V, I, & SOC values won't be modified!
13:32:07.566 > Send command: 0x90 Checksum = 0x7D
13:32:07.566 > Calculated checksum: 93, Received: 255
13:32:07.566 > Error: Checksum failed!
13:32:07.566 > RX Buffer: [,0x238,0x121,**0x255,0x255,0x255,0x255,0x255,0x255,0x255,0x255,0x255,0x255,0x255]**
13:32:07.566 > Receive failed, V, I, & SOC values won't be modified!
13:32:08.566 > Send command: 0x90 Checksum = 0x7D
13:32:08.566 > Calculated checksum: 176, Received: 255
13:32:08.566 > Error: Checksum failed!
13:32:08.566 > RX Buffer: [,0x133,0x0,0x0,0x117,0x83,0x0,0x238,0x121,0x255,0x255,0x255,0x255,0x255]